v4rl-ucy/ccm_slam #4: is it easy to change monoslam->rgbdslam?

kimwin2, 2019-01-22:
hi, 
Nowadays, we are trying to build multi-robot SLAM system.
Our usage is based on RGBD sensor, which is different from ccm_slam(mono slam).
Do you think is it easy to convert CCM SLAM from monoslam to RGBD slam?

Thanks. 
I really appreciate your affort.

patriksc, 2019-01-23:
I think it should be possible to keep the same structure and architecture for RGBD-SLAM. However, it would be necessary to adapt the modules and structures (keyframes, messages, optimization etc.) to the RGBD approach. I would assume you need to put some work in the implementation, but CCM-SLAM could provide a good starting point that you can extend towards RGBD.

cai199626, 2019-10-21:
@kimwin2 I want to know your progress of rgbd-ccm_slam, we also want to do the same thing. If you succeed,  please let me know! or there are some difficulies? Thanks a lot!

closed 2021-12-17 by patriksc

liyuting-123, 2022-08-09:
@kimwin2
I am learning this knowledge. I also want to know your progress of rgbd-ccm_slam. If you succeed, please let me know! or there are some difficulies? Thanks a lot!