v4rl-ucy/ccm_slam #6: Is it possible for server to save map?

kimwin2, 2019-02-12:
Hi, 
I really thank you for your reply.
I have one more question.

ccm slam can save map on server?
and can we load map on client?

I want to make a system which can save map on server and load map on client.

I reallly thank you for your reply.
thanks.

patriksc, 2019-02-14:
Hi 

At the moment, there isn't any functionality like this implemented. However, some time ago I was testing a function to save and load maps on the server, that then could be merged with other, new maps during the mission.

What do you want to do with the loaded map on the client side? The loading functionality was designed for the server, not for the client to re-localize against this map. However, depending on what you want to achieve, this could maybe serve as a starting point for you, that you can adapt.

Best
Patrik

arindamsaha81, 2019-04-11:
Hello,

Can you please provide me the source code for the function to save the map on the server?

patriksc, 2021-12-17:
Function available in devel branch

closed 2021-12-17 by patriksc