v4rl-ucy/ccm_slam #7: How to compute error?

ganechean, 2019-04-23:
Hi,
I'm interested in Multi-robot SLAM  system,and your repo is very helpful to me.Thank you very much.

I did some changes,and I get my own KF trajectory.But I don't know how to compute the error between my KF trajectory and Euroc MAV groundtruth when I set two agent,could you please give me some guidance? 

Thanks a lot.

SiuKeungm, 2019-04-28:
Maybe the rpg_trajectory_evaluation tools will work.

patriksc, 2019-05-02:
CCM-SLAM produces a collaborative trajectory/map of all agents. So if you run e.g. two agents using EuRoC-MH1 and EuRoC-MH2, you get a combined trajectory estimate of the two agents. For the ground truth, you can just combine the data for both datasets. Then you can calculate the error for the collaborative estimate in the same way you calculate it for only one agent/dataset, by finding correspondences between the trajectory poses and ground truth based on timestamps, then perform a Sim(3) alignment, and finally calculate the RMSE. Please note that it is important to do a Sim(3) and not a SE(3) alignment, since monocular SLAM cannot estimate the metric scale of the map/trajectory.

manthan99, 2020-06-23:
I want to calculate the error for the whole trajectory of the agents. Is there any way to write all the KFs of a particular agent into a file? I interpreted that only the KFs available at the time just after the global bundle adjustment are written to a file. 
For example, if I have two agents. Now a GBA is performed when the two maps are merged. Assuming that there is no more loop closure after this event of map merging, how do I get all the KFs of each of these agents? 
Help would be much appreciated @patriksc

patriksc, 2021-12-17:
Functionality to write KFs to file is provided by the `Map` class (`WriteStateToCsv`), and can be controlled via parameter `WriteKFsToFile`

closed 2021-12-17 by patriksc