v4rl-ucy/ccm_slam #8: Error when server and client start

arindamsaha81, 2019-06-04:
Hi,
I am getting below error when I run the server

[ccmslam/ccmslamServerNode-6] process has died [pid 18211, exit code -11, cmd /home/arindam/ccmslam_ws/devel/lib/ccmslam/ccmslamServerNode /home/arindam/ccmslam_ws/src/ccm_slam/cslam/conf/ORBvoc.txt __name:=ccmslamServerNode __log:=/home/arindam/.ros/log/6456d802-8699-11e9-a8c0-283a4d456d6f/ccmslam-ccmslamServerNode-6.log].
log file: /home/arindam/.ros/log/6456d802-8699-11e9-a8c0-283a4d456d6f/ccmslam-ccmslamServerNode-6*.log

And getting below error in case of client

[ccmslam/ccmslamClientNode0-2] process has died [pid 14051, exit code -11, cmd /home/arindam/ccmslam_ws/devel/lib/ccmslam/ccmslamClientNode /home/arindam/ccmslam_ws/src/ccm_slam/cslam/conf/ORBvoc.txt /home/arindam/ccmslam_ws/src/ccm_slam/cslam/conf/vi_euroc.yaml __name:=ccmslamClientNode0 __log:=/home/arindam/.ros/log/b6f85bfa-868d-11e9-a8c0-283a4d456d6f/ccmslam-ccmslamClientNode0-2.log].
log file: /home/arindam/.ros/log/b6f85bfa-868d-11e9-a8c0-283a4d456d6f/ccmslam-ccmslamClientNode0-2*.log

Is this issue with ccm-slam code or ROS problem? Can you please help me on this?

arindamsaha81, 2019-06-05:
Hi,

I have re-installed the entire system and now the ccm-server is running properly and client is able to start without error (I got the message on the terminal "Clientsystem initialized (Client ID: 0)") but as soon as I start playing the rosbag, the client create the map (on terminal "New Map created with 107 points") but then after immediately i got the below error on the terminal

[ccmslam/ccmslamClientNode0-2] process has died [pid 15089, exit code -11, cmd /home/arindam/ccmslam_ws/devel/lib/ccmslam/ccmslamClientNode /home/arindam/ccmslam_ws/src/ccm_slam/cslam/conf/ORBvoc.txt /home/arindam/ccmslam_ws/src/ccm_slam/cslam/conf/vi_euroc.yaml __name:=ccmslamClientNode0 __log:=/home/arindam/.ros/log/7e51f3c0-870e-11e9-8bcb-283a4d456d6f/ccmslam-ccmslamClientNode0-2.log].
log file: /home/arindam/.ros/log/7e51f3c0-870e-11e9-8bcb-283a4d456d6f/ccmslam-ccmslamClientNode0-2*.log

Please help on this..

patriksc, 2019-06-07:
Hi

Generally, it's a bit hard to guess from the information you provide what the problem is. I would recommend building e.g. in 'RelWithDebInfo' mode and using a debugger to find out exactly where the code crashes.

However, I assume in your case it might be the optimization in line 414 of Tracking.cpp. Maybe you can verify this. Please make sure you have built and linked g2o correctly. Furthermore, I experienced a similar error once I was building the code with several optimization flags, such as '-march=native'. Make sure you don't use those.

Best
Patrik

arindamsaha81, 2019-06-08:
Hi,

Thanks a lot for your reply. I really appreciate your effort.

You are correct that at line 414 at optimization in Tracking.cpp. I have built g2o but with many warnings, I hope that should not be the problem, I have attached the log file of g2o warning. ccm_slam also built with many warnings and I have attached the log for that also. I checked with the server and client node linkage with ldd command and that shows properly linked with g2o library. So I expect it is picking up the correct g2o library at run time.

I have added debug comments in Optimizer.cpp at the end of function GlobalBundleAdjustemntClient (at line 35) and at the end of function BundleAdjustmentClient (at line 211) (screenshots are attached.). Now I am getting the print from line 211 but not from line 35 just before error (attached the screenshot).

These observations may help to locate the problem. I am new to ROS but I have previous experience for debugging with gdb for C++ code and I am not sure if gdb is usefull here and in that case how to run gdb with a launch file.

Regards,
Arindam

[ccm_salm_build_log.txt](https://github.com/VIS4ROB-lab/ccm_slam/files/3268820/ccm_salm_build_log.txt)
[g2o_build_log.txt](https://github.com/VIS4ROB-lab/ccm_slam/files/3268821/g2o_build_log.txt)
[image]
[image]
[image]

patriksc, 2019-06-09:
Hi Arindam

Thanks for the further detail on the problem. By the way, which Ubuntu version are you using?

If I see that correctly from your logs, you're building with C++17, no C++14 as in our provided CMakeLists.txt. Have you tried C++14? I do not get those warning your log shows when I build the code. If possible, please try this and let me know whether it helps

I think the advantage from gdb would be to pinpoint the exact line where the code crashes. I still assume it's somewhere in the g2o library. Maybe you can also 'disable' the optimization by commenting out line160+161 in Optimizer.cpp, and see whether the it still crashes. However, without the BA, CCM-SLAM won't initialize properly, but you can see whether it's the optimization itself that causes the crash. Plus, regarding the warnings from g2o, I would check whether Eigen is installed and linked correctly.

Regarding the warning for CCM-SLAM, I would do as the log suggests: "use ‘-faligned-new’ to enable C++17 over-aligned new support": https://stackoverflow.com/questions/49373287/gcc-over-aligned-new-support-alignas

Best
Patrik

arindamsaha81, 2019-06-11:
I am using ubuntu 16.04.6 LTS and g++ 7.4.0.
I think g++ 7.4.0 version automatically enable C++17. Can you please let me know which g++ version you have used, I shall try to use the same version.

I guess the problem is not with g2o library because I have commented out the below 2 lines //optimizer.initializeOptimization();
//optimizer.optimize(nIterations);
And the error still occurring.

I have run with gdb and the log is attached. This might give some hint for finding error.
[GDB_log.txt](https://github.com/VIS4ROB-lab/ccm_slam/files/3274660/GDB_log.txt)

I would enable "use ‘-faligned-new’" and re-check.

arindamsaha81, 2019-06-11:
I have tested after enabling '-faligned-new' which stop giving those warning at compilation but still the same error occurs when run with gdb. So I guess some problem with malloc.c file.

patriksc, 2019-06-11:
Thanks for the details and further tries. I am using CCM-SLAM on Ubuntu 16 with gcc 5.4.0. and Ubuntu 18 with gcc 7.4.0., both works fine for me.

Regarding your gdb-output: you need back-trace the error to pinpoint the line where the error occurred: https://ftp.gnu.org/old-gnu/Manuals/gdb/html_node/gdb_42.html. From your log, we can only see that there is a segmentation fault, but not where it happened.

arindamsaha81, 2019-06-11:
attached is the backtrace result
[GDB_log1.txt](https://github.com/VIS4ROB-lab/ccm_slam/files/3276238/GDB_log1.txt)

patriksc, 2019-08-21:
As you can see from the log, the error comes from g2o, and as the log suggests, there seems to be a problem with the Eigen Library. I would propose to first make sure you are using the same Eigen version throughout your system. You could try to print the include paths when compiling CCM-SLAM and g2o and check whether there are two different Eigen versions used.

arindamsaha81, 2019-08-22:
I followed your library versions and that worked for me. Thanks for the info.

closed 2019-08-22 by arindamsaha81

flyover-26, 2021-12-17:
@arindamsaha81  hi!
could you please tell me what library you follow author?
thank you!