v4rl-ucy/ccm_slam #9: I notice there is no relocalization function, how client continue after lost?

mtau87, 2019-07-23:


patriksc, 2019-08-21:
Unfortunately, as you have pointed out, there is no relocalization function implemented at the moment. If the client is lost, it will not process any further incoming frames, you need the re-start the client node in this case.

matancor99, 2019-11-15:
Is there a specific reason you decided to remove the relocalization function from the original orb slam implementation? 
This make the algorithm less resilient for tracking loss (which happens quiet alot in real life).
As part of my project I am working with the implementation, and I wondered if this would be logical to add this functionality.

patriksc, 2019-11-18:
Hey, so we mainly removed the re-localization function to limit the complexity of the system, with the goal of adding this functionality at a later stage (potentially in a more sophisticated approach exploiting the centralized setup). Especially from a practical viewpoint, it would make definitely make sense to add a re-localization functionality, and the ORB-SLAM functionality would probably be a good starting point.

closed 2021-12-17 by patriksc