v4rl-ucy/ccm_slam #10: Frequently getting EstablishInitialConnectionsClient error

arindamsaha81, 2019-08-05:
Hi,
I am trying to convert ccm slam to work with stereo camera. I have done all (i think so) necessary changes to work for stereo camera and I was testing on EuRoC data set, but I am getting frequently the below error message to the client terminal.

!!!!! FATAL ERROR !!!!! EstablishInitialConnectionsClient309Constructing MP & is already associated to KF
KF-ID: 105|0 --- this MP-ID: 6468|0 --- comp. MP-ID: 6468|0
(pMP = shared_from_this()): 0

Can you please provide me the probable reasons for getting such error.

Another problem I am also facing is that the client gets track lost very frequently. Is there any specific reason for such case?

patriksc, 2019-08-21:
Hi Arindam 

Sound great, I would be very interested to see how CCM-SLAM works in a stereo case!

Regarding the error: This happens if you have two MapPoints (MPs) with the same ID - in this case 6468. And, as the check (pMP == shared_from_this()) tells us, it is not the same pointer, which means there is some inconsistency in the system. Maybe the pointer was supposed to be deleted from the map, but the removal from this KF was forgotten (you could e.g. check whether the mbBad-flag of the MP is set, which indicates that it was removed from the map).

Regarding the tracking loss: Does this happen on all sequences, especially all the MH sequences? We have tested a lot with these sequences and it works usually... However, with increasing local map size, the robustness of the system also increases. And I would propose to check whether the "Inlier Thresholds" from Tracking still work for you, since I suspect that in the stereo case, you have generally less MPs in the system.

Best
Patrik

patriksc, 2021-12-17:
Closed - inactive

closed 2021-12-17 by patriksc