v4rl-ucy/ccm_slam #14: Client - Server Communication

matancor99, 2020-01-11:
Hello,
I wanted to test the client ability to navigate in the map held by the server. 
For that I created a small experiment in which I fly two drone clients in different parts of the room, and make them merge on a certain area. After that I wanted to test the ability of one client to navigate in the regions mapped by the other client. 

I add some modifications in the code that when the merge occurs, the client stops his local mapping logic. I looked in the code and read the article, and it seems as if the client should receive MPs and KFs from the server, that are locally connected to his current one, so I though that might be a nice expirement to show that the server map is really usefull for "blind tracking".

However, my client didn't seem to learn new KFs and MPs from the server (although, by judging the communication logic in the code, it should receive the locally connected MPs and KFs and the messages), and quickly lost tracking without the local mapping actually gaining the new KFs from the Tracking thread.

My question is, do you think the experiment should work? should one client, after being merged in the server, be able to "get" KFs and MPs from the server and using them perform tracking? 
If not, how do you recomment to test the usefullness of the data in the server after the merge? 

Thank,
Matan

patriksc, 2020-01-21:
Hi Matan

Thanks for sharing your ideas! First of all, I think in general this is a great idea to test the capabilities of the system and have an effective test of map re-use. From my point of view, it should be generally feasible, however, I think you need to make some modifications in the code:

1) As it is implemented now, the server only sends a small portion of the server map to the agent, which is intended to augment the local map, but not as a stand-alone map for localization. You would have to change the logic which data is sent to the agent in a way that the full server map is shared with the agent (or at least a reasonably large part, depending how large your server map is). I would recommend to visually check this with RVIZ by coloring the parts received from the server to make sure you have enough map data around the area where the robot is currently moving that can be used for tracking.

2) There is currently no "localization-only"-mode implemented, I am not sure how stable the tracking is when you stop inserting additional KFs. Maybe you have already modified this accordingly - otherwise, you could take some inspiration from ORB-SLAM, such a functionality is implemented there, it should easily be transferable to CCM-SLAM.

I hope this helps you with your experiment - feel free to get back to me, I would be interested to see the outcome of your experiment!

Best
Patrik

patriksc, 2021-12-17:
Closed - inactive

closed 2021-12-17 by patriksc