v4rl-ucy/ccm_slam #15: Run two clients on the same dataset.

linyayunyayun, 2020-02-28:
Hey,

I ran two clients one after the other, using the same dataset(MH_01_easy). After the two maps on the server are merged, a lot of loops will be generated. As a result, the pose of the keyframes will be greatly disordered after the loop optimization. What could be the problem? Is there a solution?

[image]

Thank,
LinYayun

joegeng, 2020-04-16:
Hey,
I have the same problem as you, and i think the reason may be wrong connections between keyframes from different agents, but i can not find a solution to solve it!
Please pay attention to this problem.
Thanks!
Joe

lesurp, 2020-05-20:
Hi @linyayunyayun , I'm interested in what the problem could be. Do you mean that you run one agent on `MH_01_easy`, and once the sequence is finished, you start it again on another agent?
Or do you mean that you start the same sequence on two different agents at the same time?

linyayunyayun, 2020-05-21:
@lesurp Actually I mean that I start the same sequence on two different agents at the same time.

patriksc, 2020-05-30:
Hi LinYayun

We experience the same issue, feeding the exactly same data (i.e. the same dataset) two times to the place recognition module is causing the place recognition algorithm to produce inaccurate estimates. Is there a specific reason why you need to use the same dataset twice, and cannot use e.g. MH1 & MH2? If you really need to use MH1 twice, you could try to manually suppress some of the loops if there are to many, that might help. Or, as a quick fix, you could compare the loop transformation to ground truth data and reject it if it is incorrect. Since you would never have exactly the same data twice in a real-world scenario, this might be a quick and working approach for the development process with datasets, until the issue is fixed.

buenos-dan, 2020-07-07:
has any body compiled this project in docker?  I got some error, please help me in #23

patriksc, 2021-12-17:
Closed - inactive

closed 2021-12-17 by patriksc