v4rl-ucy/ccm_slam #17: (consistent) crashes when running on same dataset with some delay

lesurp, 2020-05-25:
After seeing #15, I did some tests, and running something such as:

```bash
rosbag play MH_01_easy.bag --start 55
rosbag play MH_01_easy.bag --start 45 /cam0/image_raw:=/cam0/image_raw1
```

Consistently crashes on agent 1 (i.e. the one who starts earlier in the sequence, and thence re-visits what agent 0 has seen)

The crashes occur in g2o because of inconsistencies in the graph, but there are at least two places that create such inconsistency

1. src/MapPoint.cpp:263
I don't know where the crash comes from exactly here
```
+++++ Enforcing KF upper limit +++++1.5904e+09
+++++ Enforcing KF upper limit +++++1.5904e+09
+++++ Enforcing KF upper limit +++++1.5904e+09
!!!!! FATAL ERROR !!!!! EstablishInitialConnectionsClient263Constructing MP & is already associated to KF
KF-ID: 56|1 --- this MP-ID: 8168|0 --- comp. MP-ID: 8168|0
(pMP = shared_from_this()): 0
!!!!! FATAL ERROR !!!!! EstablishInitialConnectionsClient263Constructing MP & is already associated to KF
KF-ID: 54|1 --- this MP-ID: 8168|0 --- comp. MP-ID: 8168|0
(pMP = shared_from_this()): 0
!!!!! FATAL ERROR !!!!! EstablishInitialConnectionsClient263Constructing MP & is already associated to KF
KF-ID: 55|1 --- this MP-ID: 7753|0 --- comp. MP-ID: 7753|0
(pMP = shared_from_this()): 0
!!!!! FATAL ERROR !!!!! EstablishInitialConnectionsClient263Constructing MP & is already associated to KF
KF-ID: 55|1 --- this MP-ID: 7968|0 --- comp. MP-ID: 7968|0
(pMP = shared_from_this()): 0
!!!!! FATAL ERROR !!!!! EstablishInitialConnectionsClient263Constructing MP & is already associated to KF
KF-ID: 54|1 --- this MP-ID: 7968|0 --- comp. MP-ID: 7968|0
(pMP = shared_from_this()): 0
!!!!! FATAL ERROR !!!!! EstablishInitialConnectionsClient263Constructing MP & is already associated to KF
KF-ID: 54|1 --- this MP-ID: 7781|0 --- comp. MP-ID: 7781|0
(pMP = shared_from_this()): 0
+++++ Enforcing KF upper limit +++++1.5904e+09
addVertex: FATAL, a vertex with ID 48 has already been registered with this graph
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
[ccmslam/ccmslamClientNode1-2] process has died [pid 15723, exit code -6, cmd /ccmslam_ws/devel/lib/ccmslam/ccmslamClientNode /ccmslam_ws/src/ccm_slam/cslam/conf/ORBvoc.txt /ccmslam_ws/src/ccm_slam/cslam/conf/vi_euroc.yaml __name:=ccmslamClientNode1 __log:=/ros/log/63b491ee-9e69-11ea-aa72-1c1b0da2ba18/ccmslam-ccmslamClientNode1-2.log].
log file: /ros/log/63b491ee-9e69-11ea-aa72-1c1b0da2ba18/ccmslam-ccmslamClientNode1-2*.log
```

2. In other cases, g2o throws a `std::bad_alloc` without the `FATAL ERROR` shown above. In this case, the crash occurs at `src/Optimizer.cpp:423` (`addVertex: FATAL, a vertex with ID 63 has already been registered with this graph`).

patriksc, 2020-05-30:
#15

patriksc, 2020-05-30:
#15

patriksc, 2020-06-05:
As mentioned in #15 , the system is not intended to feed in the exactly same data twice, this is causing some problems for the place recognition. Regarding your issue: do you experience similar problems with different dataset, say MH1 and MH2?

patriksc, 2021-12-17:
Closed - inactive

closed 2021-12-17 by patriksc