v4rl-ucy/ccm_slam #19: How to compile it in melodic?

Zcyyy, 2020-06-15:
Hi,
My environment is ubuntu 18.04 with ros-melodic.When i try to compile use catkin build,it always crash at 58%.I don't know what happened.I have tried several times.

manthan99, 2020-06-16:
Hey @Zcyyy 
I just tried out compiling with 18.04 and ros-melodic. Mine is able to compile. Could you please post the exact error which you are getting.
Just did a small change in the CMakeList of cslam package in line no. 16. Just replace kinetic with melodic inline 16.
`elseif($ENV{ROS_DISTRO} MATCHES "melodic")`

patriksc, 2020-06-16:
@manthan99 Thanks for sharing your experience.
@Zcyyy I've also just tested it on my machine, following exactly the manual - I don't have any issues compiling, apart from the check against ROS kinetic or indigo. I've adjusted the CMakeLists accordingly, should be fine now. 
Did you make sure you were following all of the steps? E.g. that the workspace is properly initialized and extended to ROS melodic, DBoW2 and g2o are built, etc.?

Zcyyy, 2020-06-17:
@manthan99 @mchli 
Thanks for sharing your experience.
I have compilied the code successful! I think the problem should be in my compiler.The compiler version I used before is based python3. when i change it,everything works well.

closed 2020-06-17 by Zcyyy