v4rl-ucy/ccm_slam #27: Why doing local mapping separately in client and server?

oym1994, 2020-08-25:
Hi,
I am very impressed in ccm-slam.  I have a question: in my point of view, clients do local mapping and then send local keyframes and key local mappoints to server. The server doesn't need to do local mapping.  But in your system,   local mapping  is separately  executed in client and server. I am very curious about the reason. 
Thanks for your attention. I am looking forward to your response!
Best regards,
Ouyang

patriksc, 2020-09-08:
Dear Ouyang
CCM-SLAM is using the same underlying class for the mapping thread on the client and server side. However, client and server call different main functions of this class (RunServer / RunClient), and perform different operation. Server Mapping mainly performs Mappoints and Keyframe culling and routes incoming data from the agents to other modules, e.g. Loop Detection and Map Matching. Client Mapping performs other tasks, such as Mappoint Triangulation and Local Bundle Adjustment, that are not done on the server side.

patriksc, 2021-12-17:
Closed - inactive

closed 2021-12-17 by patriksc