v4rl-ucy/ccm_slam #28: UAV and UGV with CCM SLAM

RodanT102, 2021-03-16:
Hello, i have a question: "how can i use your ccm slam for collaborative work of uav and ugv?" 
Does it matter to use only UAV, like in your work, or not?
Thanks for your answer

manthan99, 2021-04-29:
Hi

I do not think there are any limitations on the platform that can be used. The approach would even work with a hand-held device (like a video recorded with the help of mobile phone) since it only requires a video stream as the input.

patriksc, 2021-05-27:
Yes, as already stated above, CCM-SLAM is generally agnostic to the platform. We have even tested it already in heterogeneous teams with ground and aerial robots. The main point to consider in this setup is the viewpoint tolerance of the place recognition system - the more different the viewpoint on the scene is between 2 agents, the more difficult it is for CCM-SLAM to merge map data from those agents.

patriksc, 2021-12-17:
Closed - inactive

closed 2021-12-17 by patriksc