v4rl-ucy/ccm_slam #29: Problem in running slam on client.

Haris-Ali007, 2021-04-27:
Hello. 
I was trying to execute ccm slam on my local machines. I was successfully able to connect client with server. I made all necessary updates required to use my pc's usb cam with the code. No when I launch my usb cam it doesn't show any thing. 

Kindly guide me in solving this issue.

Haris-Ali007, 2021-05-23:
Solved it.

closed 2021-05-23 by Haris-Ali007

banciuadrian, 2021-08-13:
If I may ask, how did you go about fixing this?

banciuadrian, 2021-08-13:
If I may ask, how did you go about fixing this issue?

Haris-Ali007, 2021-08-15:
@banciuadrian I changed the order of executing my files. I used to first execute client and then my usb cam. When I switched their order, it started working perfectly fine. I did a lot of debugging to figure out why it does that but was unable to find a good supporting logic.

flyover-26, 2021-12-17:
@Haris-Ali007 Hi!
I test this system with d435i.And I just run one agent.And I found it very easy to tracking:lost if I move camera a little bit faster.
Have you got the same problem?How do you fix it?
thank you!Looking forward to your reply!