v4rl-ucy/ccm_slam #30: Facing error while installing CCM SLAM on Rasberry Pi 3b using ubuntu 18.

Haris-Ali007, 2021-06-14:
Hello. I wanted to mention an error we were encountering while installing ccm slam on rasberry pi 3b using ubuntu 18. We have been trying continuously but every time this issue occurs and disrupts the installation process. 

Kindly guide me on how we can solve this problem. 

[image]

patriksc, 2021-06-25:
Hi Haris
Hard to say what exactly is the problem, since there is no error message from the compiler. 

What I see is that the output says "g2o: Ubuntu 16 configuration", although you are reporting to use Ubuntu 18. I'd not assume this to be a problem, we have introduced this check since there was a line on g2o that needed to be changed between 16 and 18 in order for g2o to compile - so if the compiler is not throwing an error, it should most likely be fine.

However, I have an idea what could be the issue: CCM-SLAM is not such a small package and needs quite some resources to build. I experienced similar unclear build failures in the past with machines with little RAM and no Swap set up. What I'd recommend is, check your RAM utilization during compilation (e.g., using htop) - if the RAM runs full and then this error occurs, it's most likely the issue. Setting up a Swap if not existing, or extending it if running full, might solve the issue in this case.

Best
Patrik

ans-qureshi, 2021-06-30:
> Hi Haris
> Hard to say what exactly is the problem, since there is no error message from the compiler.
>
> What I see is that the output says "g2o: Ubuntu 16 configuration", although you are reporting to [...]

Hi @patriksc  I tried this again and got similar issue, this time the error is specified. And I have also added a swap memory of 4GM in addition to 1GB of my Raspberry Pi

```
In file included from /home/srl/ccmslam_ws/src/ccm_slam/cslam/include/cslam/Optimizer.h:47:0,
                 from /home/srl/ccmslam_ws/src/ccm_slam/cslam/include/cslam/MapMatcher.h:52,
                 from /home/srl/ccmslam_ws/src/ccm_slam/cslam/include/cslam/Mapping.h:40,
                 from /home/srl/ccmslam_ws/src/ccm_slam/cslam/include/cslam/ClientHandler.h:47,
                 from /home/srl/ccmslam_ws/src/ccm_slam/cslam/include/cslam/CentralControl.h:38,
                 from /home/srl/ccmslam_ws/src/ccm_slam/cslam/include/cslam/Communicator.h:52,
                 from /home/srl/ccmslam_ws/src/ccm_slam/cslam/include/cslam/MapPoint.h:41,
                 from /home/srl/ccmslam_ws/src/ccm_slam/cslam/include/cslam/KeyFrame.h:40,
                 from /home/srl/ccmslam_ws/src/ccm_slam/cslam/src/KeyFrame.cpp:25:
/home/srl/ccmslam_ws/src/ccm_slam/cslam/thirdparty/g2o/g2o/solvers/linear_solver_eigen.h:45:50: note: #pragma message: g2o: Ubuntu 16 configuration
 #pragma message ( "g2o: Ubuntu 16 configuration" )
                                                  ^
/home/srl/ccmslam_ws/src/ccm_slam/cslam/src/KeyFrame.cpp:31:18: error: conflicting declaration ‘size_t cslam::KeyFrame::nNextId’
 size_t KeyFrame::nNextId=0;
                  ^~~~~~~
compilation terminated due to -Wfatal-errors.
make[2]: *** [CMakeFiles/ccmslam_lib.dir/src/KeyFrame.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/ccmslam_lib.dir/all] Error 2
make: *** [all] Error 2
cd /home/srl/ccmslam_ws/build/ccmslam; catkin build --get-env ccmslam | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << ccmslam:make                [ Exited with code 2 ]               
Failed    <<< ccmslam                     [ 12 hours 51 minutes and 23.0 seconds ]
[build] Summary: 1 of 2 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  1 packages succeeded with warnings.                       
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 12 hours 51 minutes and 29.0 seconds total. 
```

jamesdi1993, 2021-07-31:
I'm facing a similar issue here with Ubuntu 18 and ROS Melodic.

patriksc, 2021-08-03:
@ans-qureshi Thanks for the follow-up.If I'm not mistaken, the error has changed now to an explicit error message, not to an unidentifiable termination of the build process, right?

Looking at the code, I saw that in "KeyFrame.h", _KeyFrame::nNextId_ is declared as `static long unsigned int nNextId;`, while in "KeyFrame.cpp", it is defined as `size_t KeyFrame::nNextId=0;`. This is clearly not optimal, since `size_t` and ` long unsigned int` since `size_t` is system-dependent, while `long unsigned int`is fixed, and I suspect this causes the reported error.

@ans-qureshi @jamesdi1993 I would therefore ask you to unify both lines, e.g. by replacing  `size_t KeyFrame::nNextId=0;` in `KeyFrame.cpp` with `long unsigned int KeyFrame::nNextId=0;`, and post here whether this solved the problem. Thanks!

patriksc, 2021-12-17:
Closed - inactive

closed 2021-12-17 by patriksc