v4rl-ucy/ccm_slam #31: Error in trying to launch any launch file.

laibazahid26, 2021-06-30:
Hello,
I built this project successfully. But when I try to launch the server I get the error below:
[image]

This error does not tell anything much about the issue. I don't understand where to look to debug it. 
I am using Ubuntu 18.04 and the processor is AMD Ryzen 7. I have 12 GBs of memory.

closed 2021-07-02 by laibazahid26

ans-qureshi, 2021-07-04:
Hi @laibazahid26 I am facing the same problem, how did you resolve this?
[image]

laibazahid26, 2021-07-04:
Hi, 
I think you are launching the Client in the wrong way. You need to type 
**roslaunch ccmslam Client0_euroc.launch**
inorder to launch Client 0. 

However, my problem was different. I still do not understand the reason for this error. But I uninstalled Ubuntu and installed Docker (on my Windows) on my laptop and installed Ubuntu image on Docker. And everything ran perfectly.

laibazahid26, 2021-07-04:
> Hi @laibazahid26 I am facing the same problem, how did you resolve this?
> [image]
> [image]

Hi,
I think you are launching the Client in the wrong way. You need to type
roslaunch ccmslam Client0_euroc.launch
inorder to launch Client 0.

However, my problem was different. I still do not understand the reason for this error. But I uninstalled Ubuntu and installed Docker (on my Windows) on my laptop and installed Ubuntu image on Docker. And everything ran perfectly.

ans-qureshi, 2021-07-04:
@laibazahid26 no, I renamed the launch file. Anyways, i'll try some other way. Thanks!

emily-68, 2022-03-21:
> Hello, I built this project successfully. But when I try to launch the server I get the error below:
> 
> [image]
> 
> This error does not tell anything much about the issue. I don't understand where to look to debug it. I am using Ubuntu 18.04 and the processor is AMD Ryzen 7. I have 12 GBs of memory.

Hello, I also encountered this problem. I modified CMakeLIst,This problem is due to the OpencV version,You need to modify the file"/home/siat/twq_ws/ccm_slam/src/ccm_slam/cslam/CMakeLists.txt",  find_package(OpenCV 3 REQUIRED)，Please do not use OPencv 2.4.13