v4rl-ucy/ccm_slam #32: Error contacting the email under License section in the ReadMe

srispoli-metis, 2021-07-07:
Hi, I am interested in acquiring the commercial version of CCM-SLAM so I sent an inquiry to the email listed in the License section of the ReadMe. However the email server cannot find the address. I verified that I was writing to the correct email with the papers listed. I tried with 2 different domains, one through Outlook and one through Gmail and got the same error code both times. Is there an alternate address I can contact to inquire about the commercial version? Thanks.

patriksc, 2021-07-08:
Hi, thanks for reaching out. The listed mail address has unfortunately expired, and we haven't updated it yet - thanks for pointing this out, and sorry for the inconvenience. The new point of contact for the repo is "collaborative.slam@gmail.com", and we would kindly ask you to contact this address for your request. We will update the repo information as well in the next days. Thank you.

srispoli-metis, 2021-07-09:
Great, thanks Patrik!

closed 2021-07-09 by srispoli-metis