v4rl-ucy/ccm_slam #33: How do I know which set of keypoints correspond to which image frame?

laibazahid26, 2021-09-06:
Hello @mchli , @patriksc

Is there any way of knowing which set of keypoints correspond to which image frame?

According to my understanding, some of the image frames are treated as keyframes, and from these keyframes, the keypoints are being generated. 

I am working on an image stitching (panorama building) problem. For this purpose, I want to get keypoints and descriptors from your repository and stitch images based on these keypoints and descriptors. 
At the moment, I am subscribing to topic '/ccmslam/MapOutClient0' and extracting. This topic has type Map (your builtin) message. From this message when I extract: 
your_map_message.Keyframes[0].mnId, 
the mnIds I get apparently do not correspond to keyframes sent. 
So, if mnids = 13, this does not correspond to image 13.

In short, If let's say I extract all the keypoints from the message 'map.msg', how can I know which set of keypoints correspond to which image frame? Can you please tell me how do I extract this information?

patriksc, 2021-09-14:
Hello Laiba

Each KF stores the timestamp of the image it is was created from (the field `mTimeStamp`). With this timestamp, you should usually be able to uniquely identify the image (for example, for the EuRoC dataset, the timestamp is identical to the image name, such as "1403636860801666560.png")

Hope that helps!

laibazahid26, 2021-09-14:
> Hello Laiba
>
> Each KF stores the timestamp of the image it is was created from (the field `mTimeStamp`). With this timestamp, you should usually be able to uniquely identify the image (for example, [...]

I see but, I have to extract keyframes from the 'map.msg' because that's the only published message which contains the information about keyframes too.  According to my understanding, mappoints are generated from keyframes but when I extract keyframes from mappoints, some of the array of keyframes come out to be empty. How is this possible?

laibazahid26, 2021-09-15:
This helped. Thanks a lot.

closed 2021-09-15 by laibazahid26