v4rl-ucy/ccm_slam #34: problem about euroc dataset

flyover-26, 2021-11-09:
hi!
I test euroc dataset on one PC and I play rosbag at a half speed.
And I got result like this.
I find that if I play only two dataset at the same time,I got a map with two diffenrent color.
[image]

But If I play four rosbags at the same time,I got a point cloud map with only white point and MarkerC0,MarkerC1,MarkerC2 and MarkerC3 got an error like below.Is this necessary to fix?
[image]

And I got an error like Tracking: lost in client2.
Could you tell me how to fix it?
Thank you so much!

flyover-26, 2021-11-09:
And somtimes I got a problem like this?
Does this matter?

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

+++++ KeyFrame Database Initialized +++++
Map 3 rand seed: 1636425490
+++++ Map 3 Initialized +++++
Camera Input topic: /cam0/image_raw3
Client 3 --> Initialize Threads

Clientsystem initialized (Client ID: 3)
[ INFO] [1636425490.391562312]: Started CSLAM client node...
New Map created with 68 points
Wrong initialization, reseting...
System Reseting
Reseting Done...
New Map created with 133 points
+++++ Enforcing KF upper limit +++++1.63643e+09
+++++ Enforcing KF upper limit +++++1.63643e+09
+++++ Enforcing KF upper limit +++++1.63643e+09

patriksc, 2021-11-18:
Hi! Regarding the `Enforcing KF upper limit` - this means the front-end get did not get an acknowledgement from the server that the sent KFs have arrived, but the local map is getting too large, so it is telling you it is anyway now removing old map data inside the local map. Does the data arrive on the serve side, i.e. can you see the map data that this agent has shared? If not, most likely there is something wrong with the topic names. If the data arrives, maybe the system is just overloaded, what could be the case when you are playing four rosbags simultaneously.

Regarding the visualization: The error displayed in RVIZ do not matter, we do not initialize the quaternion of the marker messages, since we are not using the rotation entry for visualization. _TopicName_C* are trajectories & point clouds from the individual agents, you are probably looking for the data on the server side, like in the first image. Either the data did not arrive (e.g., because the agent lost track), or the maps did not merge (you can check that in the console output), or something changed in the visualization, and the point cloud is not color coded anymore (check the color coding of MapPointsS* in RVIZ)

Regarding the tracking loss of MH2: if it works with 2 agents, but not with 4, probably you system is overloaded. Maybe try running the bagfiles at a lower playback speed, using `rosbag play mybag.bag -r 0.25`. 

Overall, I would recommend to try running you experiment with first with 3 and then with 4 agents, and using `rosbag play mybag.bag -r 0.25`, and post here some feedback how this worked, so that we get more insights what might be the issue.

patriksc, 2021-11-18:
Also, we have recently released our follow-up system, COVINS: https://github.com/VIS4ROB-lab/covins. However, it works with visual-inertial data, not on monocular image streams.

flyover-26, 2022-01-03:
thank you for your reply!
I have another question is that now I use two pc to do this project.
I run agent0 on pc1.And I run agent1 and server on pc2.
The bandwith of wifi is about 800KB/S.
And I also got Enforcing KF upper limit.Is this problem related to the bandwith of internet?or someother reason?
thank you for your patience!

patriksc, 2022-01-06:
The high bandwidth most likely comes from many KFs inserted into the map. This can be because of your motion pattern, or due to a higher framerate of your camera compared to the EuRoC sequence where we have evaluated CCM-SLAM. You can try to tackle the issue by trying out different parameters for the communication, as [described in the readme](https://github.com/VIS4ROB-lab/ccm_slam#6-parameters). Besides that, increasing the values for `Mapping.LocalMapSize` and `Mapping.LocalMapBuffer` might help when you run into `Enforcing KF upper limit`.

flyover-26, 2022-02-12:
thank you so much for your help.
when I test this system on dataset I got a result like this.It is obvious that the  bule trajectory is wrong.

[image]
what I focus is that when use my own camera,I go around the room,everthing goes well.But I will get the same problem like blue trajectory above after the system run several global BA.(use only one agent) Is there any reason leading to this problem?

The second question is that I found if I use two agents,after the map merged happening,global BA and closed loop will happen frequently when two agents pass by the the place ( it takes a lot of time).Is this normal phenomenon？Because I use orb-slam2 to test same scene,global BA and closed loop rarely happen.

Thank you for your patience!

patriksc, 2022-03-06:
Thanks for sharing your results. Apart from the blue trajectory, I would also say the white trajectory is wrong - it's not 100% clear to tell from the image, but usually start and end should align, it looks like there is some significant drift. 

Since you are sing EuRoC, which should usually produce reasonable results, I suspect your system might not have enough resources available to run CCM-SLAM properly - e.g. running front- and back-end simultaneously on a PC with not much compute power might result in the estimate diverging.

Regarding Loop Closure / BA: Yes, it is possible that CCM-SLAM finds loop closures very frequently if the scene allows it. Each loop closure detection (LCD) is then usually followed by global BA, which might take a while to finish. BA can be interrupted by e.g. a subsequent LCD, however, the current iteration will be finished. But you can influence the frequency of LCD: the parameter `Placerec.NewLoopThres` in `config.yaml` specifies by how many KFs tow consecutive LCDs must at least be separated.

flyover-26, 2022-03-07:
thank you so much for your help!
I use two pc.one is server and agent1 another is agent2.
And I test this system in a room.I find sometime after BA the trajectory will become strange.
this result is the good one(I tested several times and only this time I get a good result)
[image]

But most of time (map merged with no problem),when find closure loop and BA happend,it seems somethings wrong(It should have almost same trajectory as above  ).And I have no idea what is going on?
[image]

patriksc, 2022-03-13:
Thanks for sharing those details. Regarding your results (particularly the ¨good¨ result): although sparse maps only represent a small amount of the 3D structure of the true environment, it's usually possible to relate them to the real scene and identify e.g. corners of tables, walls, etc. I do not know your environment, but your 3D points seem rather unstructured from the image you shared, for the fact that this environment is an indoor room (I assume an office / lab environment). I would suspect that already the front-end estimate might be inaccurate/incorrect, and for this reason, also BA/Loop closure fails, because the input data is flawed. This can of course not definitely be said from a remote perspective, however, I would recommend to check the estimate of your front-end (deactivate BA and LC), and compare the estimated 3D point cloud with the real environment, to make sure this is correct.

patriksc, 2022-07-31:
Problem seems resolved - closing the issue.

closed 2022-07-31 by patriksc