v4rl-ucy/ccm_slam #36: A problem about your paper

WangJiaheng1, 2021-12-08:
I noticed that you use an additional module,which you called  "communication controller", in your paper CCM‐SLAM ,to measure the communication bandwidth for different ploss and delay settings. How did you implement this part, or is it included in your open source code? thanks!

patriksc, 2021-12-13:
The communication controller is unfortunately not part of the open-source code. It was implemented using a separate ROS Node, which subscribed to the individual topics published by server and agent, and published them again under a different name. However, e.g. in the case of message loss with a chance of 0.2, it would drop a message with 20% probability, so that on average only 4 out of 5 messages would be published again.

To make an example, if you would want to introduce message loss for the data that agent 0 send to the server (topic: `MapOutClient0`), then you would have a separate ROS Node that subscribes to `MapOutClient0`, and with e.g. a probability of 80%, publishes the exact same message again under the topic `MapOutClient0Disturbed`. The server would then subscribe to `MapOutClient0Disturbed`, and not to `MapOutClient0` as set right now.

WangJiaheng1, 2021-12-14:
I got it.Thank you for your kind reply!

closed 2021-12-14 by WangJiaheng1