v4rl-ucy/ccm_slam #37: problem about traking lost.

flyover-26, 2021-12-14:
hi!
I use d435i camera to test ccm-slam on one Xavier NX.
And I found it's easy to get traking lost if I move d435i a little bit fast.
I want to know if there is any reason to lead to this problem?
Is this related to fps ?(my d435i's fps is 30)
Thank you very much!

flyover-26, 2021-12-14:
And I want to use on NX with d435i to let the map merged happen.
What I do is run server node and agent0 node,and I can get a map.And the I shut down agent0 node and run agent1 node on the same nx.And I also get a map.However,It seem that map merged didn't happen.
I want to know is this method right to get merged map?
thank you!

flyover-26, 2021-12-14:
[image]
[image]

patriksc, 2021-12-17:
Hi! 
- Regarding your first problem (frequent tracking loss): This might come from the limited CPU compute power of the Xavier NX. CCM-SLAM is not optimized for GPUs, and I think the CPU power of the Xavier NX is significantly less compared to e.g. the Intel NUCs we would usually use. It should be sufficient though, as long as you don't create a too challenging setup. Particularly, avoiding fast motions, textureless areas, and few scene depth (i.e. wall close to the camera) has shown to reduce tracking loss when running CCM-SLAM on limited compute.
- Regarding point 2 (failed map merge): Yes, this procedure should work. You can even double check with the EuRoC dataset, if the procedure works with this, it should also work with your datasets. Most likely, you don't have enough view overlap to merge the maps, or map quality is not good enough - try to have an area with a similar view on the environment, this should help.
- Another point to mention, I would recommend to double-check your camera calibration, and whether you are using the right distortion model (CCM-SLAM expects radtan). Feeding an inaccurate camera calibration to CCM-SLAM will result in bad performance of the pipeline, creating problems like bad map quality, tracking loss, or failed merges.

flyover-26, 2021-12-18:
Thank you  for your reply!
Now I use d435i to test this project on pc.
And I just run one agent on pc,when I open rviz to subcribe /ccm-slam/ImageClient0 the agent0 crash like this.
[image]
By the way,I use ubuntu 18.04.

And I notice that you have a devel vision,does this has any difference comparing with master one?
Thank you!

flyover-26, 2021-12-18:
I found If I set camera'resolution to 1280*720 ,it can run nomally.
But if I set it to 640*480,it will crash.

patriksc, 2021-12-18:
As mentioned in the readme `For the newest version of CCM-SLAM (1.1), including functionalities to save and load maps, try out the devel branch`

flyover-26, 2021-12-18:
thank you so much  for your reply!
Could you tell me when you do experienment with 3 UGV in real world,what resolution does your camera have?Will the stability of system be related to the resolution of camera?or the vision of ubuntu? like the problem above?

Is ccm-slam(1.1) for ubuntu 18.04?
thank you so much for your patience!!!

patriksc, 2021-12-18:
CCM-SLAM should work well with Ubuntu 18. Generally, a higher camera resolution can improve the quality of the SLAM estimate, but will also affect the timings. Both resolutions you mentioned should work (EuRoC has 752x480). However, be aware that different resolution will require different camera parameters (fx,fy,cx,cy, maybe even distortion coefficients) - so if you just change e.g. the resolution of the cam to 640x480 and not the camera parameters, this will fail.

flyover-26, 2021-12-20:
THANK YOU !!!
Now I can let two maps merged, but I can not see red lines as you described.(ps:What I do is run server node and agent0 node,and I can get a map.And the I shut down agent0 node and run agent1 node on the same nx.And I also get a map.)

[result.txt](https://github.com/VIS4ROB-lab/ccm_slam/files/7742586/result.txt)

In this result.txt,it seems map merged happened uncorrectly.

[image]

Could you tell me is this result correct?
I appreciate your help!

patriksc, 2021-12-20:
Covisiblity edges (red and gray) are only displayed if they have a weight above a certain threshold. This can be set via the parameter `Viewer.CovGraphMinFeats` - probably your edges are not strong enough. 
Whether this results is correct I cannot tell, since I neither know your camera motion nor the structure of your environment. What I can say is, the white and green point cloud seem to align (not really clear to say from the perspective though), which would be a good sign.

flyover-26, 2021-12-23:
thank you so much for your help!
About resolution,I use gdb to debug this problem,and get error below.
Could you give some advice on this?
[image]

flyover-26, 2021-12-23:
[image]

flyover-26, 2021-12-23:
[image]

patriksc, 2021-12-23:
Not exactly clear what is going on from those logs - can you please try to spot the exact line that causes the segfault? Also, probably introduce a check whether `mCurrentFrame` in `Viewer::UpdateFrame` is valid or a `nulltpr`.

flyover-26, 2021-12-27:
that is the exact line I can provide.
But sometimes,the program goes well with resolution 640x480.Most time it will crash with 640x480.
Is this the problem with project or with d435i?
I am clined to d453i to lead to this problem,because everying goes well with 1280x720.And do also good with 640x480 sometimes.Is this project sensitive to camera input?
Thank you so much!!!

Accidently,I run nx as an agent0 and pc as agent1 、server.
When map merged happened,I got a merge map with only white point.It seems not stable enough.And sometimes,I got a merged map with two diffenent color which is correct.
Is there any reason about this?
thank you so much for your help!

flyover-26, 2021-12-27:
[image]

all white point like this.

flyover-26, 2021-12-28:
I got the exact line that has error.
Hope this will be helpful!
Thank you so much for your patience.

[gdb.txt](https://github.com/VIS4ROB-lab/ccm_slam/files/7783551/gdb.txt)

patriksc, 2022-01-03:
- In the example image you provided, the maps seem to have merged, so you should have 2 different colors for the 3D points. Maybe some preference in RVIZ has changed, e.g. the `Channel` of the PointCloud topic is set to `Intensity`. 

- An image resolution if 640x480 should definitely work, as long as you have the correct camera parameters. Are you sure you are not accidentally using the same camera parameters for both resolutions? This would well explain why it works well for 1280x720 but is unstable for 640x480.

- Regarding your segfault: From your `gdb` output, I assume `mInitialFrame` in line 329 of `Viewer.cpp` is a `nullpointer`. You could as well verify this to be sure. Since this error seems to be specific to your machine / setup, and I cannot replicate it, I cannot provide a straightforward solution, you would need to find the root cause on your machine. As a quick fix, you could e.g. try to comment out line 147 in `Tracking.cpp`: `mpViewer->UpdateAndDrawFrame();` - you will not see the images until the system is initialized, but the segfault should not appear anymore.

patriksc, 2022-07-31:
Problem seems resolved - closing the issue.

closed 2022-07-31 by patriksc