v4rl-ucy/ccm_slam #39: problem about experiment

flyover-26, 2022-01-05:
hi!
I test ccm-slam in realworld,and I found a problem.
When I move camera to a new scene,I found sometime it can not extracting feature points.I will track lost if I continue to move camera.
Is this related to nFeatures in configure.yaml?or other reasons?
thank you for your help!
[image]
(ps:in the left side of picture,it can extract feature points,because it's the scene it has experienced.I move camera from left to right)

patriksc, 2022-01-06:
Those are not the extracted features, but the ones that have a 3D point associated. Most likely, the front-end could not insert a new KF yet, and use it to triangulate new 3D points on the right side of the scene. I think you mentioned you are using a Xavier NX - maybe the CPU computation power is not sufficient, which can result in degraded performance with effects like this. Another reason could be too swift motion. Try moving slower, and to increase the distance between camera and scene if possible, this might help.

flyover-26, 2022-01-07:
thank you so much for your help!
I  successfully let map merged happened with two agents.But when I continue to run the system,I got a warning like !!! WARN !!! Replace:663 id mismatch MP 24895
--- optimize Ess grapgh
--- Launch GBA thread
LoopFinder: Wait for GBA to finish
-> Starting Global Bundle Adjustment
--> Optimizing Map 1 -- Contains Agents 0;1;
----> Loop KF: 154|0
----- Add KFs
----- Add MPs 
And it is obvious that there is something wrong with map like trajectory distortion.
Is this related to mismatch?(I change nFeatures in configure.yaml to 1500)or other reasons?
I appreciate your help!

patriksc, 2022-01-18:
A single mismatched landmark should have a discernible impact of the estimate. A trajectory distortion could be caused by a wrong set of camera parameters. Particularly, if you observe a large plane perpendicular to the camera viewing direction, and the 3D points on this plane appear bent, this is a clear indicator of wrong camera parameters (e.g. the ground plain for a downward looking camera, but as well a large wall in a hand-held setup).

patriksc, 2022-07-31:
Problem seems resolved - closing the issue.

closed 2022-07-31 by patriksc