v4rl-ucy/ccm_slam #43: problem about mapmatch and mapmerge

lantianran, 2022-04-04:
Hello, I have successfully run the program, and I am more concerned about the closed-loop module between multiple machines. I found at runtime that the system always performs MapMatch first, and then MapMerge. If there are two clients, it will only be performed once, and if there are three clients, it will only be performed twice. Isn't the closed loop between multiple agents being detected all the time? , it should be more than that little. 
From the run log it looks like all agents only did one MapMatch.
@mchli @patriksc 
Hope to get your help！

patriksc, 2022-04-14:
Hello lantianran
Actually, loop closures between multiple agents are continuously detected in CCM-SLAM. I think there might be some ambiguity regarding the used terminology: When two separate maps _M0_ and _M1_ from two different agents _A0_ and _A1_ are **matched** and subsequently **merged**, this generates a new map (let's call it _M01_), which is now shared by both agents _A0_ and _A1_. Any further successful place recognition for this map is then treated as a **loop closure**, since it operates on the same map, even when the involved data comes from different agents.

flyover-26, 2022-06-16:
I noticed that there are a lot variables  like g2oS_c_wm.Could you please explain the  general meaning of these variables?
I have no idea what these variables can do? thank you so much.
 ps:
g2o::Sim3 g2oS_c_wc(Converter::toMatrix3d(Rcw),Converter::toVector3d(tcw),1.0); //cam - world client
 g2o::Sim3 g2oS_c_wm = g2oS_c_wc*(mg2oS_wcurmap_wclientmap.inverse()); //cam client - world map
@patriksc

patriksc, 2022-06-26:
This is the 7DoF Sim(3) transform type provided by g2o (Rot, Trans, Scale). `g2oS_c_wc` means the Sim(3) transform from the world coordinate frame of the client into it's camera coordinate frame, i.e. right-multiplying a point in the world coordinate system transform it into the camera coordinate system.

patriksc, 2023-12-31:
Closed - inactive

closed 2023-12-31 by patriksc