v4rl-ucy/ccm_slam #44: static transform publisher

flyover-26, 2022-05-20:
hi!
I notice that there is a static transform publisher parameter. I think when map merges,we can get every agent pose which means these is no need to know their initial pose set by static transform publisher .
Whether this parameter is only for visualization in rviz?And when map merge,this parameter become useless?
thank you for your help~

patriksc, 2022-05-20:
Yes, the static transform publisher is only for visualization purposes. When maps are merged, some of the transforms might not be used anymore, but that's not a problem.

flyover-26, 2022-05-31:
I got it!Thank you so much.
I notice in this project MapPoint_complicated.cc is not included in cmakelists.Is this file for any purpose?And I found a frame constrcutor in MapPoint_complicated.cc  which can be used to create temporary MapPoint in tracking thread.(used in stereo or rgbd  in UpdateLastFrame() form TrackWithMotionmodel() in orb2)
Is it right if I use this constructor directly to create temporary MapPoint if I want to modify ccm to stereo version?just like orb-slam2?thank you!!!!

patriksc, 2022-06-11:
This file most have slipped the refactoring efforts - it is some deprecated piece of code from earlier stages of development. Please ignore it, we will also remove it in one of the next commits then. I would definitely discourage using this code.

patriksc, 2022-06-11:
Removed via b2b5d6e2700869133e3d9f34119d7a5da91d758c

patriksc, 2023-12-31:
Closed - resolved

closed 2023-12-31 by patriksc