v4rl-ucy/ccm_slam #47: problem about parameters configuration

flyover-26, 2022-06-22:
hi!I noticed in your paper the number of KFs in local map is set to 50 and parameters k (number of KFs sent from server to agent to augment local map) is set to 5.
However,in config.yaml ,I noticed that Comm.Server.KfsToClient is set to 0.
Should  this parameter  set to 5? 
Looking forward to you reply!

patriksc, 2022-06-26:
Which value you choose for this parameter depends a bit on the bandwidth of your wireless connection, but you can set it to 5, yes. We might have set it to 0 it for some experiment in the past.

flyover-26, 2022-07-12:
@patriksc 
Thanks! I change ccm to stereo-version.However, I notice some red points appear in current frame.Is this a bad signal that I made some mistake?I find in ccm or orb-slam2 no red points or blue points appear in current frame.
Thank you so much for your help.
[image]

flyover-26, 2022-08-01:
thank you for your reply.
When I set this parameter to 5 in stereo version ,I got a problem.
(
!!!!! FATAL ERROR !!!!! EstablishInitialConnectionsClient335Constructing MP & is already associated to KF
KF-ID: 99|1 --- this MP-ID: 52054|1 --- comp. MP-ID: 52054|1
(pMP = shared_from_this()): 0

)
I am not sure why this hanppend.Could you please offer some reasons?thank you so much.

By the way,will this parameter has a huge difference to system's stability if set to 5?

patriksc, 2023-12-31:
Do you get the same problem when using the mono version? CCM-SLAM is meant to be used in monocular mode, we have never tested with the stereo version.

patriksc, 2024-04-28:
Closing for now  - pleas re-open if the issue is still present.

closed 2024-04-28 by patriksc