v4rl-ucy/ccm_slam #49: Camera Pose in world reference

DanielRoeder1, 2022-07-05:
I am trying to perform 3D reconstruction using the depth maps of the TUM datasets in conjunction with the poses of CCMSLAM. 
I know that the camera poses of CCMSLAM are not scaled correctly but I take care of this separately. 

Now to perform this reconstruction correctly from my understanding I would need the Twc (camera pose in world reference) to transform the point clouds resulting from the depth maps. But how would I get the camera pose so that it aligns correctly with the typical xyz coordinate frame? 

Currently, the poses are coherent to a somewhat twisted coordinate frame and then transformed  by this static transformer:
https://github.com/VIS4ROB-lab/ccm_slam/issues/44

I found this issue in the ORBSLAM2 repo describing how to correctly visualize the poses:
https://github.com/raulmur/ORB_SLAM2/issues/832

While the approach from the ORBSLAM2 repo seems to help the poses are still not fully aligned.

Could someone give a hint about the transformation that is needed or some explanation on the transformation between the ORB frame and world frame?

patriksc, 2022-07-31:
H Daniel, is this still a problem, or did you already solve it (sorry for the late reply here)? If yes, could you maybe share a bit more information where you are stuck, I'm not sure whether I can exactly follow what actual problem is here. Particularly, is your dense 3D reconstruction from depth data globally consistent, but just shifted/twisted w.r.t. the sparse 3D points? Or are the dense points clouds correct withing a single camera frame, but not correctly aligned, so that the 3D reconstruction itself is inconsistent? If you could maybe also share an image, that would probably also help.

patriksc, 2023-12-31:
Closed - inactive

closed 2023-12-31 by patriksc