v4rl-ucy/ccm_slam #50: How can I add the number of agents to 8 or more?

fengshiyu58, 2022-07-11:
I would like to add the number of agents to 8 to compare CCM-SLAM with our methods, but I found it was difficult to do this. Is there a close source edition CCM-SLAM that support 8 or more Agent? 
By the way, I noticed your latest work on COVINS, but it need IMU information and I need a monocular algorithm.
Looking forward your reply, Thanks!

patriksc, 2022-07-31:
Hi, there is unfortunately currently no version of CCM-SLAM implemented that supports to use more than 4 agents. However, the choice of max. 4 agents is only an implementation choice, CCM-SLAM can technically support also more agents, and it should be quite easy to extend the code to do this.  You would have to adjust the launch files (agent / server / static transforms) to support more agetns, and add the infrastructure for agents 5-8 in `ServerSystem`.

fengshiyu58, 2022-08-01:
Thanks a lot for your patient reply, I will try to do this.

pengllbit, 2022-09-07:
Have you solved this problem now.

> Thanks a lot for your patient reply, I will try to do this.

Have you solved this problem now. I want run ccmslam with 5 agents.

pengllbit, 2022-09-07:
can you share the source code for me to learn?

patriksc, 2023-12-31:
Closing - inactive

closed 2023-12-31 by patriksc