v4rl-ucy/ccm_slam #51: Memory consumption is too large

flyover-26, 2022-07-23:
hi！I test ccm on kitti and I noticed that when dataset run for 3/4  it need memory about 9g while orb-slam2 only cost 3g.I have no idea why it need so much memory(It should not be related to the number of KF ,because it has almost the same number of KF as orb-slam2).And it need more than 20g when I change it to stereo-version which is to large for my computer.
 Is there any reason may result in this?Looking forward to your reply!

patriksc, 2022-07-31:
Hi! We have not in particularly analyzed the memory footprint of CCM-SLAM, so I cannot say exactly why you see a significant difference compared to ORB-SLAM. I would probably recommend to quickly use a profiler to see whether there are some memory hotspots that can be easily reduced.

patriksc, 2023-12-31:
Closed - inactive

closed 2023-12-31 by patriksc