v4rl-ucy/ccm_slam #55: cannot run client and server on two PC

JoannaZhj, 2022-09-12:
Hi, I have followed the steps and it's OK to perform both client and server on one PC. After I use two PCs and execute the ROS_MASTER_URI in terminal, client performs well. But server cannot show the map. Do I miss something? I'm sure that two PCs connect to the same wifi.
Thanks for your help!

patriksc, 2022-10-29:
Maybe you also have to set the local ROS IP as well with `export ROS_IP=<your-local-IP>` (in each terminal, use on each PC local IP)
I would recommend to follow a tutorial implementing a simpler  interaction than a full collaborative SLAM system here, to make sure everything is OK and correctly set up. If that works, but you still don get a connections between the agents and the server, please let us know. Here's some pointers to tutorials:
- https://wiki.ros.org/ROS/Tutorials/MultipleMachines
- https://github.com/husarion/static_docs_deprecated/blob/master/src/core2/tutorials/ros-tutorials/5-running-ros-on-multiple-machines.md

JoannaZhj, 2023-01-16:
OK, I'll give it a try.Thank you very much.

patriksc, 2023-12-31:
Closed - inactive

closed 2023-12-31 by patriksc