v4rl-ucy/ccm_slam #57: Map merge results

JazzyFeng, 2022-11-26:
First thanks for your excellent work. I run Server backend on my own desktop to merge three agents trajectory on EuRoC dataset. However, sometimes the graph is distorted. Maybe the loop closure edge is wrong? The RMSE ATE is around 0.20 for MH_01 and MH_02, respectively.

patriksc, 2022-12-30:
Hi, thanks for you feedback. Overall, we have experienced the results on the EuRoC dataset as quite stable, but it can of course happen that occasionally, something goes wrong, and the resulting estimate is not as accurate as usually (e.g. due to a wrong loop closure, as you mentioned). How often does this happen? Do you run the agents on the same desktop, in parallel? Generally, one source for higher error we have seen is if the PC does not have enough compute power to run the server as well as multiple agent front-ends in parallel, could this be the case in your setup?

patriksc, 2023-12-31:
Closed - inactive

closed 2023-12-31 by patriksc