v4rl-ucy/ccm_slam #58: Load error！

sun1f, 2023-02-07:
Hi, I am so impressed with this amazing cslam project. 

I got a little question here: 
Aftering successfully saving maps by using rosservice command, I set the parameter `LoadMap` in `Server.launch` to true in order to load maps, jsut following the guidance of `README`. But something went wrong, is there anything else I should do when loading maps?

Hope to receive your reply！
[image]

patriksc, 2023-02-11:
Hi, thanks for your message. It looks like something has gone wrong here, the map you are trying to load contains a landmark with an ID not known to the map. Could be a glitch in the function that saved the map. Did you try this with multiple maps, and the error is reproducible over multiple saved/loaded maps? Or did it just happen for a single map? 

For this specific map, you could try changing the code to just skip this landmark in the loop and continue with the next one, instead of directly exiting. if it's only a single landmark with this error, the map might still work. But generally, stored maps should be able to be loaded again.

patriksc, 2023-12-31:
Closed - inactive

closed 2023-12-31 by patriksc