v4rl-ucy/ccm_slam #60: Enforcing KF upper limit

junglej, 2023-03-06:
I have two machines which can be called Client and Server.
And client agent communicated with the server. But when the client sent KF to the server, the client had this log.
[image]

Could you give some tips about this log? What do they mean?

Config:
Client: Ubuntu 18.04
Server: Ubuntu 20.04

patriksc, 2023-03-17:
This message is displayed when the client does not get confirmation from the server that KFs were received, and needs to delete KFs without this confirmation at some point. Are you using the default parameters? 
If I understand you correctly, the data arrives at the server, and you see the trajectory of the client displayed there, right? (otherwise, this would explain the message). Apart from that, a reason could be that your WiFi network does not have enough bandwidth and the data piles up on the client side over time - could that be possible? 
What you could try in this case is play a bit with the parameters - for mapping, `Mapping.LocalMapBuffer`, and the Communication parameters [mentioned in the readme](Mapping.LocalMapBuffer).

patriksc, 2023-12-31:
Closed - inactive

closed 2023-12-31 by patriksc