v4rl-ucy/ccm_slam #61: map visualization

chocho2222, 2023-04-06:
Thank you very much for your contribution, I ran this program successfully. But I'm a novice in slam. I didn't find the map created by visualization in rviz. I couldn't find the corresponding map topic by using "ros topiclist". Can you guide me how to visualize it? Thanks again。

patriksc, 2023-04-16:
Hi, to make sure everything is working fine, I would recommend to run all nodes (agent and server) on the same machine for the start. Then, in a new terminal, follow exactly the steps described here: https://github.com/VIS4ROB-lab/ccm_slam#41-examples-on-the-euroc-dataset. Did you maybe forget to source `setup.bash` in the terminal you are running the visualization from?

patriksc, 2023-12-31:
Closed - inactive

closed 2023-12-31 by patriksc