v4rl-ucy/ccm_slam #62: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'

TwiceMao, 2023-05-15:
I meet this error，how could I solve it？

[ccmslam/ccmslamClientNode1-2] process has died [pid 999291, exit code -6, cmd /media/zrr/B6DC2D32DC2CEE75/maomao/ccmslam_ws/devel/lib/ccmslam/ccmslamClientNode /media/zrr/B6DC2D32DC2CEE75/maomao/ccmslam_ws/src/ccm_slam/cslam/conf/ORBvoc.txt /media/zrr/B6DC2D32DC2CEE75/maomao/ccmslam_ws/src/ccm_slam/cslam/conf/replica_vi_euroc.yaml __name:=ccmslamClientNode1 __log:=/home/zrr/.ros/log/991423c8-f2e8-11ed-b628-ed28cb613320/ccmslam-ccmslamClientNode1-2.log].
log file: /home/zrr/.ros/log/991423c8-f2e8-11ed-b628-ed28cb613320/ccmslam-ccmslamClientNode1-2*.log

[image]
[image]

patriksc, 2023-05-27:
Hi! According to the error message, you are using OpenCV 4. We have only tested CCM-SLAM with OpenCV 2 and 3, and since the error comes from an OpenCV call, that's very likely the root cause here. I would recommend to link against OpenCV 3, that should solve the problem.

patriksc, 2023-12-31:
Closed - inactive

closed 2023-12-31 by patriksc