v4rl-ucy/ccm_slam #64: Data packet from server to client about client's past experience when visiting a mapped location was not found.

Jufer1, 2023-05-29:
The system was run on the EuRoC dataset with four clients enabled. I collected the server to client packet MapOutServer and did not see any past experience with the mapped data in it. The packet looks like this:

---
header: 
  seq: 0
  stamp: 
    secs: 1685193487
    nsecs: 664260356
  frame_id: ''
mMsgId: 1
Keyframes: []
KFUpdates: []
MapPoints: []
MPUpdates: []
vAckKFs: [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12]
vAckMPs: [1, 3, 5, 7, 11, 17, 18, 20, 22, 24, 31, 37]
WeakAckKF: 65535
WeakAckMP: 4294967295
ClosestKf_Id: 0
ClosestKf_ClientId: 0
---

Where can I find the mapped past experience packet that the server sends to the client?

patriksc, 2023-06-09:
Hi, did you change the parameter `Comm.Server.KfsToClient`? (cf. https://github.com/VIS4ROB-lab/ccm_slam#6-parameters). It's set to 0 by default (https://github.com/VIS4ROB-lab/ccm_slam/blob/b054ef343d9d99fcc695d8ca525594d0466a5a49/cslam/conf/config.yaml#L96).

patriksc, 2023-12-31:
Closed - inactive

closed 2023-12-31 by patriksc